hybridgroup/cylon-raspi #30: Project location for download

adonisc, 2015-01-08:
Hi! i was about to install the cyclon-raspi framework but fails showing 404 error this thru NPM and on github. If you can provide the file again thanks in advance! by the way im downloading thru an RPi

deadprogram, 2015-01-08:
Hi @adonisc seems fine on https://www.npmjs.com/package/cylon-raspi perhaps you were having a network problem?

adonisc, 2015-01-08:
Yes it seems the package there but it was throwing that it was marked as
moved permanently and/or not found errors (403,404) i will try again today.

Thanks for the quick answer!!!

adonisc, 2015-01-09:
Hi, i don't know if i'm doing anything wrong but can't download thru npm it continues throwing 404 error :(

deadprogram, 2015-01-09:
Is your raspi able to access other npm packages?

deadprogram, 2015-01-09:
can we get a look at your package.json? or are you just using the npm command directly?

adonisc, 2015-01-09:
Thanks Ron i already manage the problem i already using cylon! Thanks a lot
for your help.

deadprogram, 2015-01-09:
Glad to hear it!

closed 2015-01-09 by deadprogram