hybridgroup/cylon-raspi #32: npm install cylon-raspi fails

ColorfullyMe, 2015-04-14:
Hi have some issue installing the modlue no my Pi2. What coud be reason for this:

```
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/opt/node/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.EventEmitter.emit (events.js:101:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:827:12)
gyp ERR! System Linux 3.18.7-v7+
gyp ERR! command "node" "/opt/node/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/dev/cytest/node_modules/cylon-raspi/node_modules/i2c
gyp ERR! node -v v0.11.9
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm ERR! Linux 3.18.7-v7+
npm ERR! argv "node" "/opt/node/bin/npm" "install" "i2c"
npm ERR! node v0.11.9
npm ERR! npm  v2.7.6
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/dev/cytest/node_modules/cylon-raspi/npm-debug.log
npm ERR! Linux 3.18.7-v7+
npm ERR! argv "node" "/opt/node/bin/npm" "install" "cylon-raspi"
npm ERR! node v0.11.9
npm ERR! npm  v2.7.6
npm ERR! code ELIFECYCLE

npm ERR! cylon-raspi@0.16.0 install: `./scripts/install-i2c.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the cylon-raspi@0.16.0 install script './scripts/install-i2c.sh'.
npm ERR! This is most likely a problem with the cylon-raspi package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ./scripts/install-i2c.sh
npm ERR! You can get their info via:
npm ERR!     npm owner ls cylon-raspi
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/dev/cytest/npm-debug.log
```

Any help will be appreciated!
Thanks

deadprogram, 2015-04-14:
I see you are running Node.js version `0.11.9`. I've not tried that, the most recent prebuilt binary for Raspi ARM I've installed is `0.10.28`.

I followed the install instructions here: https://github.com/hybridgroup/cylon-raspi#update-your-raspbian-and-install-nodejs and had no problems on the Pi2.

I see there are Pi builds for http://nodejs.org/dist/v0.11.7/ thru http://nodejs.org/dist/v0.11.12/ (they stop after that for some reason) so I will try one of those versions to see if I can repro your problem.

ColorfullyMe, 2015-04-14:
@deadprogram Hi i have downgraded node to 10.9 and it works with it. Just wanted to inform you...Apparently there is something wrong with this package and node > 0.11 Please. close the issues as you find appropriate. Thanks for quick response!

deadprogram, 2015-04-14:
This is related to https://github.com/kelly/node-i2c/issues/16 I think

bchr02, 2015-12-30:
I am having the same issue. Why was this closed? I realize that it might be related to https://github.com/kelly/node-i2c/issues/16 but regarless it's still an open issue when installing cylon-raspi.

deadprogram, 2015-12-30:
Hi, @bchr02

deadprogram, 2015-12-30:
Oops, too quick on the click. Anyhow, @bchr02 how would you suggest we handle this? Perhaps we need to update the README/docs for cylon-raspi to warn people about version compatability issues.

bchr02, 2015-12-30:
Perhaps adding something to the README (this would definitely help) but sometimes it's nice to also have an OPEN issue too because that's the first place one would go to when seeing if it's a known issue. Thanks

deadprogram, 2015-12-30:
That seems like a good idea. I'm going to adopt that concept.

fivdi, 2015-12-30:
@deadprogram have you ever considered using an alternative i2c module to resolve the issue?

bchr02, 2015-12-30:
@deadprogram how about https://github.com/fivdi/i2c-bus?

deadprogram, 2016-01-02:
I just realized @fivdi *has* an alternative, which is the one that @bchr02 suggests. :)

No need to be so subtle, dear friends, this project is absolutely open to any suggestions.

Looks to me like like a great idea. Would one of you be willing to work on a PR that replaces it?

subjaeger, 2016-01-05:
A bit of Version testing

```bash
Test environment:
	Hardware: Raspberry Ip 2
	OS: RASPBIAN JESSIE 4.1 (2015-11-21)
	Firmware: 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux
```

```bash
npm install cylon cylon-raspi cylon-gpio cylon-i2c	-	Did it work?
----------------------------------------------------------------------------
node: v0.10.29		npm: 1.4.21								yes
node: v0.10.41		npm: 1.4.29								yes
node: v0.10.41		npm: 3.5.2								no
node: v0.12.9		npm: 2.14.9								yes
```

deadprogram, 2016-03-06:
Hi, @fivdi & @bchr02 

I have a branch here https://github.com/hybridgroup/cylon-raspi/tree/feature/better-i2c that switches over to using i2c-bus. Any feedback would be appreciated!

fivdi, 2016-03-06:
I took a look at [i2c-device.js](https://github.com/hybridgroup/cylon-raspi/blob/68e352dae37ddea110f02d6e6ebc44e1d76c6a90/lib/i2c-device.js) and the mapping of i2c to i2c-bus methods is correct.

Note that although the i2c methods look asynchronous and have callbacks, under the covers they are actually synchronous. They don't use worker threads for reading and writing from and to the I2C bus. This means that they will return AFTER the operation has been performed. The corresponding callbacks are called in the next turn of the event loop. 

i2c-bus methods are really asynchronous and return immediately without waiting for the operation to complete. They use worker threads for reading and writing from and to the I2C bus.

This synchronous/asynchronous mismatch between i2c and i2c-bus may or may not result in issues.

i2cWrite is not being called correctly [here](https://github.com/hybridgroup/cylon-raspi/blob/68e352dae37ddea110f02d6e6ebc44e1d76c6a90/lib/i2c-device.js#L30) and [here](https://github.com/hybridgroup/cylon-raspi/blob/68e352dae37ddea110f02d6e6ebc44e1d76c6a90/lib/i2c-device.js#L35), it has four arguments, not three.

The [read method](https://github.com/hybridgroup/cylon-raspi/blob/68e352dae37ddea110f02d6e6ebc44e1d76c6a90/lib/i2c-device.js#L33-L42) splits what should be one transaction into two transactions as it calls both i2cWrite and i2cRead. In theory (and in practice) another process (or the same process) could attempt to access the same device between the two calls and mess things up. Typically, readI2cBlock would be used instead as it's one method call instead of two. It's also guaranteed to be a single transaction which can't be interrupted by other calls, be they in the same process or another process. On the other hand, if the implementation with the two calls worked in the past, perhaps it's better to leave it be for the moment and maybe switch over to readI2cBlock in the future.

Is [install-i2c.sh](https://github.com/hybridgroup/cylon-raspi/blob/feature/better-i2c/scripts/install-i2c.sh) needed? Why not add i2c-bus as a dependency in package.json?

deadprogram, 2016-03-06:
Hi, @fivdi I updated my branch with both the corrections you point out, along with switching to `readI2cBlock` as you suggested. Please check it out.

deadprogram, 2016-03-06:
Regarding the script for install the reason is so that devs who on OSX can install the prereqs for development on this module, by just running `npm install`. Otherwise, the install fails on non-Linux platforms.

fivdi, 2016-03-07:
If the synchronous/asynchronous mismatch mentioned above doesn't result in any issues then it looks like i2c-device.js should do what it did before.

This code [here](https://github.com/hybridgroup/cylon-raspi/blob/feature/better-i2c/scripts/install-i2c.sh#L7-L11):
```bash
if [[ $node_version == *"v0.10."* ]]; then
  npm install i2c-bus
else
  npm install i2c-bus
fi
```
can be rewritten as:
```bash
npm install i2c-bus
```

deadprogram, 2016-03-07:
I've updated the branch to simplify the install script as you suggested, thanks @fivdi

kawashimaken, 2016-03-15:
Hope to see it merged to master soon. :-)

deadprogram, 2016-03-15:
I am planning on going some hw tests today so this can get released, thanks for being patient!

deadprogram, 2016-03-19:
OK, took a little longer, and had to make a few more code changes, but the master branch now contains the new code, that works in all version of Node.js

I will be releasing a new version later today, unless anyone raises objections. Thanks everyone for the help on this!

deadprogram, 2016-03-19:
Version cylon-raspi@0.19.0 is now released. I am closing this issue, please either reopen or open new issue(s) with any problems with the new implementation. Thanks again, everyone!

closed 2016-03-19 by deadprogram

bchr02, 2016-03-20:
Thanks @deadprogram and sorry for not being able to test it out yet. When I do get a chance, if there's anything I'll open a new issue. Good job!