hybridgroup/cylon-raspi #35: driver: 'led' toggles every start from crashing to working

subjaeger, 2016-01-05:
When I run the led demo from the git page, nodejs crashes every first run.
On the second run it works perfectly. It toggles every start from crashing to working.

I am using:
```bash
Hardware: Raspberry Ip 2
Firmware: 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux
NodeJS: v0.10.29 (Stock)
NPM:1.4.21
```
Install and run:
```bash
npm install cylon cylon-raspi cylon-gpio cylon-i2c
node cylon_test_led.js
```

cylon_test_led.js:
```javascript
var Cylon = require("cylon");

Cylon.robot({
  connections: {
    raspi: { adaptor: 'raspi' }
  },

  devices: {
    led: { driver: 'led', pin: 38 }
  },

  work: function(my) {
    every((1).second(), my.led.toggle);
  }
}).start();
```
error-msg:
```bash
node cylon_test_led.js
2016-01-05T14:06:41.748Z : [Robot 1] - Starting connections.
2016-01-05T14:06:41.804Z : [Robot 1] - Starting connection 'raspi'.
2016-01-05T14:06:41.811Z : [Robot 1] - Starting devices.
2016-01-05T14:06:41.812Z : [Robot 1] - Starting device 'led' on pin 38.
2016-01-05T14:06:41.813Z : [Robot 1] - Working.

events.js:74
        throw TypeError('Uncaught, unspecified "error" event.');
              ^
TypeError: Uncaught, unspecified "error" event.
    at TypeError (<anonymous>)
    at DigitalPin.emit (events.js:74:15)
    at DigitalPin._setModeCallback (/home/pi/Share/testDir/1_npmTest/node_modules/cylon/lib/io/digital-pin.js:152:17)
    at DigitalPin.<anonymous> (/home/pi/Share/testDir/1_npmTest/node_modules/cylon/lib/io/digital-pin.js:146:10)
    at fs.js:953:21
    at Object.oncomplete (fs.js:107:15)
```

deadprogram, 2016-01-09:
That looks to me like kind of permissions problem when first trying to create the `export` file that contains the GPIO pin mappings. Can you please run `whoami` and also `groups` commands, and provide the output of both?

subjaeger, 2016-01-09:
"whoami" -> "pi" 
"groups" -> "pi adm dialout cdrom sudo audio video plugdev games users input netdev gpio i2c spi"

deadprogram, 2016-01-09:
And do you have any udev rules setup?

subjaeger, 2016-01-11:
No, I didn't. It is basically still the stock firmware. Besides "apt-get upgrade".
The odd thing is that is is working... every 2nd start of the script.

deadprogram, 2016-01-11:
I suggest you try to follow the steps on https://github.com/hybridgroup/cylon-raspi#connecting-to-raspberry-pi-gpio to correct permissions problems. This is a common requirement on pretty much any Linux GPIO.

Please give that a try, and let us know how it goes.

deadprogram, 2016-03-19:
Hi, @subjaeger were you able to try setting the udev rules as described in the README?

deadprogram, 2016-04-27:
Hi @tbarteit were you able to get past your issue?

matamalaDDG, 2017-07-19:
hi, im follow the cylon.js LED using RPI tutorial but the program does't run, and I follow all the instruction.
 I have a RPI 3 model B, and i use this code

script.js:

```
var Cylon = require("cylon");

Cylon.robot({
  connections: {
    raspi: { adaptor: 'raspi' }
  },

  devices: {
    led: { driver: 'led', pin: 11 }
  },

  work: function(my) {
    every((1).second(), my.led.toggle);
  }
}).start();
```

so wen i try to run my programs using 

```
npm install cylon cylon-raspi cylon-gpio cylon-i2c
node script.js
```

and follow all the instruction for the installation
the program throw this error 

error-msg:
```
pi@piSystem:~/Cylon$ node script.js 
2017-07-19T04:52:37.293Z : [Robot 1] - Starting connections.
2017-07-19T04:52:37.322Z : [Robot 1] - Starting connection 'raspi'.
2017-07-19T04:52:37.326Z : [Robot 1] - Starting devices.
2017-07-19T04:52:37.326Z : [Robot 1] - Starting device 'led' on pin 11.
2017-07-19T04:52:37.326Z : [Robot 1] - Working.

events.js:74
        throw TypeError('Uncaught, unspecified "error" event.');
              ^
TypeError: Uncaught, unspecified "error" event.
    at TypeError (<anonymous>)
    at DigitalPin.EventEmitter.emit (events.js:74:15)
    at DigitalPin.<anonymous> (/home/pi/Cylon/node_modules/cylon/lib/io/digital-pin.js:117:12)
    at fs.js:944:21
    at Object.oncomplete (fs.js:107:15)
```

tnks for your time, and i wait your answer

hellnano, 2017-12-11:
did you try using `SUDO` before your commands? ( like `sudo node scripts.js` ) ... if this "solve" the problem, then you need to check your permissions .

closed 2019-01-15 by tbarteit

warlock, 2019-01-15:
Not resolved.