hybridgroup/cylon-raspi #36: throw TypeError('Uncaught, unspecified "error" event.') on LED Blink

bencarter78, 2016-01-19:
Hi 

Sorry total noob to cylon/cylon-raspi, just going through the 'hello-world' example and I get the following error...

```
2016-01-19T19:31:02.214Z : [Robot 1] - Starting connections.
2016-01-19T19:31:02.403Z : [Robot 1] - Starting connection 'raspi'.
2016-01-19T19:31:02.425Z : [Robot 1] - Starting devices.
2016-01-19T19:31:02.429Z : [Robot 1] - Starting device 'led' on pin 12.
2016-01-19T19:31:02.431Z : [Robot 1] - Working.

events.js:74
        throw TypeError('Uncaught, unspecified "error" event.');
              ^
TypeError: Uncaught, unspecified "error" event.
    at TypeError (<anonymous>)
    at DigitalPin.emit (events.js:74:15)
    at DigitalPin.<anonymous> (/home/pi/code/node_modules/cylon/lib/io/digital-pin.js:117:12)
    at fs.js:953:21
    at Object.oncomplete (fs.js:107:15)
```

And no blinking LED. What is the error telling me?

deadprogram, 2016-01-19:
Hi, @bencarter78 I would suspect a problem with GPIO and permissions. Check out https://github.com/hybridgroup/cylon-raspi#connecting-to-raspberry-pi-gpio

bencarter78, 2016-01-19:
All I had to do was run node hello-world.js with sudo and it worked!

Thanks @deadprogram 

As an aside I wanted to let you know that I found CylonJS through an podcast you did with the Changelog and because of your enthusiasm, energy and positivity. Thanks for helping me find this project!

deadprogram, 2016-01-19:
:bow:

closed 2016-01-19 by bencarter78