hybridgroup/cylon-raspi #40: DigitalPin TypeError: Cannot read property 'toString' of undefined

scottmillers, 2016-08-22:
I have a Raspberry Pi 3 with Raspbian GNU/Linux 8 (jessie) and node v6.4.0 and npm 3.10.3

I followed the cylon install instructions up to the install of pi-blaster as 
>sudo apt-get install pi-blaster 
fails due to it be not being able to be locate the package.  

When I run my code I get the following:

pi@millerspi:~ $ cat cylon.js 
var Cylon = require("cylon");

Cylon.robot({
	connections: {
		raspi: {adaptor: 'raspi'}
	},
	
	devices: {
		led: {driver: 'led', pin: 17}
	},
	
	work: function(my) {
		console.log("hello");
		every((1).second(), my.led.toggle);
	}
}).start();

pi@millerspi:~ $ sudo node cylon.js
2016-08-22T19:02:54.507Z : [Robot 1] - Starting connections.
2016-08-22T19:02:54.554Z : [Robot 1] - Starting connection 'raspi'.
2016-08-22T19:02:54.563Z : [Robot 1] - Starting devices.
2016-08-22T19:02:54.563Z : [Robot 1] - Starting device 'led' on pin 17.
2016-08-22T19:02:54.564Z : [Robot 1] - Working.
hello
/home/pi/node_modules/cylon/lib/io/digital-pin.js:25
  this.pinNum = opts.pin.toString();
                        ^

TypeError: Cannot read property 'toString' of undefined
    at new DigitalPin (/home/pi/node_modules/cylon/lib/io/digital-pin.js:25:25)
    at Raspi._digitalPin (/home/pi/node_modules/cylon-raspi/lib/raspi.js:313:29)
    at Raspi.digitalWrite (/home/pi/node_modules/cylon-raspi/lib/raspi.js:178:16)
    at Led.turnOn (/home/pi/node_modules/cylon-gpio/lib/led.js:83:19)
    at Led.toggle (/home/pi/node_modules/cylon-gpio/lib/led.js:111:10)
    at Timeout.wrapper [as _onTimeout] (timers.js:425:11)
    at tryOnTimeout (timers.js:232:11)
    at Timer.listOnTimeout (timers.js:202:5)
pi@millerspi:~ $ 

Any ideas?

Thanks,
Scott

warlock, 2017-10-16:
The same problem usign rasbian stretch
```
pi@zero:~/exptest $ node index.js 
2017-10-16T23:05:09.424Z : [Robot 1] - Starting connections.
2017-10-16T23:05:09.462Z : [Robot 1] - Starting connection 'raspi'.
2017-10-16T23:05:09.480Z : [Robot 1] - Starting devices.
2017-10-16T23:05:09.485Z : [Robot 1] - Starting device 'led' on pin 17.
2017-10-16T23:05:09.488Z : [Robot 1] - Working.
/home/pi/exptest/node_modules/cylon/lib/io/digital-pin.js:25
  this.pinNum = opts.pin.toString();
                         ^

TypeError: Cannot read property 'toString' of undefined
    at new DigitalPin (/home/pi/exptest/node_modules/cylon/lib/io/digital-pin.js:25:26)
    at Raspi._digitalPin (/home/pi/exptest/node_modules/cylon-raspi/lib/raspi.js:313:29)
    at Raspi.digitalWrite (/home/pi/exptest/node_modules/cylon-raspi/lib/raspi.js:178:16)
    at Led.turnOn (/home/pi/exptest/node_modules/cylon-gpio/lib/led.js:83:19)
    at Led.toggle (/home/pi/exptest/node_modules/cylon-gpio/lib/led.js:111:10)
    at ontimeout (timers.js:471:11)
    at tryOnTimeout (timers.js:306:5)
    at Timer.listOnTimeout (timers.js:266:5)
pi@zero:~/exptest $ 
```

warlock, 2017-10-18:
I solved the problem.  I don't remember the solution.

Maybe you can enable i2c in:
 raspi-config > Interfacing Options > I2C

And intall pi-blaster:
[pi-blaster](https://github.com/sarfata/pi-blaster)

closed 2024-01-12 by scottmillers