hybridgroup/cylon-raspi #41: Error while creating pin files

alexarroyoduque, 2017-08-08:
Hi, I'm using a raspberry pi 2B.

I have been experimenting with the CylonJS library following the documentation example. However when I run the script I get the following error.

Can anybody help me?

```javascript
2017-08-07T18:50:16.659Z : [Robot 1] - Starting connections.
2017-08-07T18:50:16.677Z : [Robot 1] - Starting connection 'raspi'.
2017-08-07T18:50:16.705Z : [Robot 1] - Starting devices.
2017-08-07T18:50:16.713Z : [Robot 1] - Starting device 'led' on pin 18.
2017-08-07T18:50:16.720Z : [Robot 1] - Working.
events.js:187
      throw err;
      ^

Error: Unhandled "error" event. (Error while creating pin files)
    at DigitalPin.emit (events.js:185:19)
    at DigitalPin.<anonymous> (/home/pi/led-project/node_modules/cylon/lib/io/digital-pin.js:117:12)
    at fs.js:1279:7
    at FSReqWrap.oncomplete (fs.js:135:15)
```

```javascript
var Cylon = require("cylon");
Cylon.robot({
  connections: {
    raspi: { adaptor: 'raspi' }
  },

  devices: {
    led: { driver: 'led', pin: 18 }
  },

  work: function(my) {
    every((1).second(), my.led.toggle);
  }
}).start();
```

warlock, 2017-10-18:
I have the same problem. Investigating and looking to see if I find the way to solve the problem.

raspberry zero w + raspbian strech

nodejs 8.7.0 + cylon 1.3.0

--------

```
pi@zero:~/exptest $ node index.js
2017-10-18T18:43:31.140Z : [Robot 1] - Starting connections.
2017-10-18T18:43:31.178Z : [Robot 1] - Starting connection 'raspi'.
2017-10-18T18:43:31.196Z : [Robot 1] - Starting devices.
2017-10-18T18:43:31.200Z : [Robot 1] - Starting device 'red' on pin 40.
2017-10-18T18:43:31.204Z : [Robot 1] - Working.
events.js:187
      throw err;
      ^

Error: Unhandled "error" event. (Setting up pin direction failed)
    at DigitalPin.emit (events.js:185:19)
    at DigitalPin._setModeCallback (/home/pi/exptest/node_modules/cylon/lib/io/digital-pin.js:152:17)
    at DigitalPin.<anonymous> (/home/pi/exptest/node_modules/cylon/lib/io/digital-pin.js:146:10)
    at fs.js:1280:7
    at FSReqWrap.oncomplete (fs.js:136:15)
```

mymattcarroll, 2017-11-07:
@warlock, @alexarroyoduque, I was having the same issue running a very similar script by simply `ssh`ing into our pi and running this script: 

```js
'use strict'

const Cylon = require('cylon')

const PIN_NUMBER = 11
let interval = null

Cylon
  .robot({ name: 'TestBot' })
  .connection('raspi', { adaptor: 'raspi' })
  .device('led', { driver: 'led', pin: PIN_NUMBER })
  .on('error', function (error) {
    console.log('got an error!!!!')
    console.error(error)
  })
  .on('ready', function(bot) {
    interval = setInterval(function() {
      console.log(`Toggling LED at pin: ${PIN_NUMBER}`)
      bot.led.toggle(function () {
        console.log('Toggled LED')
      })
    }, 1000);
  });

Cylon.start();

process.on('SIGTERM', function () {
  console.log('SIGTERM event caught')
  if (interval) {
    clearInterval(interval)
  }
  Cylon.halt(function () {
    console.log('Halted Cylon')
  })
})
```

Was only able to get passed the error by running the script using sudo:

```
sudo node index.js
```

warlock, 2017-11-08:
The project has been abandoned. It's sad because Cylon is the best known system of this type for Nodejs. I will try to find time to understand how it works and maybe I will write a fork or an alternative.

MiguelAngel82, 2020-07-29:
Hi @warlock , @alexarroyoduque  and @mymattcarroll,

I know this project was abandoned and this issue quite old, but, were you been able to solve this problem? I'm facing the very same issue, and although is not active this project, it still works fine for some mini-projects.

Thanks in advance!

Regards,
Miguel.