hybridgroup/cylon-raspi #43: always error when running with cylon-raspi modules

Wavesolid, 2022-05-24:
I have installed both Cylon and CYlon-raspi locally in some folders inside my Raspi, but when I tried to run the code it always give me an errors message "Cannot find the 'cylon-raspi' module. This problem might be fixed by installing it with 'npm install cylon-raspi' and trying again. 

As u can see in the photo that I provided, Cylon-raspi is installed inside node_modules folders.
[image]

tobiaswendling, 2022-06-12:
I have the same problem and no fix for it. Any solutions?

dangerousthing, 2024-01-06:
I have the same problem, with the same error message. This is with a new installation of 
Cylon on a Raspberry Pi. Even a hit may help.